Open-CMSIS-Pack/cpackget #283: [cpackget] fails to install the packs specified in the `requirements` section

jkrech, 2024-04-10:
**Describe the bug**
When installing the pack version `NXP::EVK-MIMXRT1064_BSP@14.0.0` using this command:
```
cpackget add NXP::EVK-MIMXRT1064_BSP@14.0.0 -v
```
the expectation is that due to these `<requirements>`

```
<requirements>
    <packages>
      <package name="MIMXRT1064_DFP" vendor="NXP" version="14.0.0"/>
      <package name="CMSIS" vendor="ARM" version="5.6.0"/>
    </packages>
</requirements>
```
that if no pack version of `NXP::MIMXRT1064_DFP >= 14.0.0` is already installed, the latest version listed in the .Web/index.pidx gets downloaded and installed.
(similarly for "ARM::CMSIS >= 5.6.0")

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure that you do not have any of the packs NXP::EVK-MIMXRT1064_BSP@14.0.0 and NXP::MIMXRT1064_DFP installed
2. `cpackget add NXP::EVK-MIMXRT1064_BSP@14.0.0 -v`
3. See error: 
```
D: Downloading https://mcuxpresso.nxp.com/cmsis_pack/repo/NXP.MIMXRT1064_DFP.18.0.0.pack to C:\Users\user\AppData\Local\Arm\Packs\.Download\NXP.MIMXRT1064_DFP.18.0.0.pack
D: Download not required, using the one from cache
I: Adding pack NXP.MIMXRT1064_DFP.
D: Installing "C:\Users\user\AppData\Local\Arm\Packs\.Download\NXP.MIMXRT1064_DFP.18.0.0.pack"
D: Validating pack
D: Inflating "NXP.MIMXRT1064_DFP.pdsc"
D: Ensuring "C:\Users\user\AppData\Local\Temp\vCGxspJRfM\" directory exists
D: Inflated 193693 bytes
D: Initializing PdscXML object for "C:\Users\user\AppData\Local\Temp\vCGxspJRfM\NXP.MIMXRT1064_DFP.pdsc"
D: Reading pdsc from file "C:\Users\user\AppData\Local\Temp\vCGxspJRfM\NXP.MIMXRT1064_DFP.pdsc"
D: Making sure  is the latest release in NXP.MIMXRT1064_DFP.pdsc
E: The latest release (18.0.0) in pack's pdsc (NXP.MIMXRT1064_DFP.pdsc) does not match pack version ""
E: pack version is not the latest in the pdsc file
```

**Expected behavior**
The expectation is that NXP::MIMXRT1064_DFP@18.0.0 gets successfully installed.

**Environment (please complete the following information):**
 - CMSIS-Toolbox 2.3.0
 - cpackget 
 - OS: windows
 - Version 2.1.1

jkrech, 2024-04-10:
Looks like a duplicate of:
https://github.com/Open-CMSIS-Pack/cpackget/issues/282

edriouk, 2024-04-10:
Two problems have identified and fixed: 
1. cpackget internally uses `"_"` as unlimited upper range, e.g. `"1.2.3:_"` 
  ` utils.SemverCompareRange()` is updated to support it. 
2. `pack.resolveVersionModifier() `: swapped version and range arguments to call ` utils.SemverCompareRange()` 
 corrected

closed 2024-04-12 by jkrech