Open-CMSIS-Pack/cpackget #357: [cpackget] Explicit installation may fail

ReinhardKeil, 2024-10-14:
**Describe the bug**

When installing a newly published pack, an error like this may appear.

```
cpackget add Keil::B-U585I-IOT02A_BSP@2.0.0
I: Adding pack "Keil::B-U585I-IOT02A_BSP@2.0.0"
E: pack version not found in the pdsc file
```

**To Reproduce**

Remove in the CMSIS_PACK_ROOT folder:
- The folder Keil\B-U585I-IOT02A_BSP
- The file .\Web\B-U585I-IOT02A_BSP.pdsc
- From file .\Web\index.pidx the `<pdsc url` reference to `B-U585I-IOT02A_BSP`.

Execute: `cpackget add Keil::B-U585I-IOT02A_BSP@2.0.0`

As a user you get this line from the web portal, and you perhaps don't know that `cpackget update-index` should be executed first.

jkrech, 2024-10-29:
When installing a public pack, the index.pidx needs to be opened.
a) if index.pidx does not exist, invoke the behavior of `cpackget init https://www.keil.com/pack/index.pidx` -> bail out on error 
b) if index.pidx exists - open index.pidx and read `<timestamp>`
c) if `<timestamp>` date is smaller than host date invoke behavior of `capckget update-index --sparse` -> bail out on error
d) now continue with the behavior of `cpackget add packID`, locating pack in index.pidx, downloading pdsc file if version in index is higher than version in pdsc file.
e) reading pdsc file `<releases>` section to find latest or specified version. If no version match found bail out. 
f) read url attribute from release tag or construct download url from elements `<url>` `<vendor>` `<name>` `version` . pack

jkrech, 2024-12-10:
Validating nightly build of CMSIS-Toolbox from 2024-12-09:
With an outdated <timestamp> in .Web/index.pidx the `cpackget add` command download the index.pidx.

```
cpackget add YTMicro::YTM32B1LD0_DFP@1.1.1
I: Respond: 200:200 OK (online)
I: Downloading index.pidx...
I: 100% |██████████████████████████████████████████████████████████████████████████████| (210/210 kB, 298 kB/s)
I: Downloading YTMicro.YTM32B1LD0_DFP.pdsc...
I: 100% |██████████████████████████████████████████████████████████████████████████████| (3.4/3.4 kB, 5.3 MB/s)
I: Adding pack "YTMicro::YTM32B1LD0_DFP@1.1.1"
I: Downloading YTMicro.YTM32B1LD0_DFP.1.1.1.pack...
I: 100% |██████████████████████████████████████████████████████████████████████████████| (209/209 kB, 240 kB/s)
I: Extracting files to c:\tmp\cmsis-packs\YTMicro\YTM32B1LD0_DFP\1.1.1...
I: 100% |███████████████████████████████████████████████████████████████████████████████████| (15/15, 407 it/s)
```
Also in case the CMSIS_PACK_ROOT folder is empty, the `cpackget add` will bootstrap the initialization of the required folders.

closed 2024-12-10 by jkrech