Open-CMSIS-Pack/cpackget #358: [cpackget] Improve robustness for handing `index.pidx` file

ReinhardKeil, 2024-10-14:
**The Problem To Be Solved**

The `index.pidx` file is the central file for pack management.  I recommend the following improvements:

- Make error message a bit more descriptive when `index.pidx` file is missing.  Currently `cpackget update-index --sparse` reports:
    ```
    cpackget update-index --sparse
    I: Updating public index
    E: file not found
    ```
    Improved message suggestion:
    ```
    E: index.pdix file not found; use cpackget init command to retrieve it
    ```
- Update `index.pidx` even when the local time stamp is newer. This may need a bit more work, as it effectivly requires a checksum or equivalent in the index.pidx file.

jkrech, 2024-10-29:
Execution steps:
a) open index.pidx, read `<timestamp>` element `date` (before to `T`) information and compare against current host date. 
b) read `<url>` element and download `<url>index.pidx` if the current date is higher than the timestamp
c) report error if the download fails due to lack of internet connection or incorrect url

jkrech, 2024-12-10:
With the above mentioned PR (see nightly build), the index.pidx is downloaded when it does not exist and the machine is online. Otherwise an error message reports the missing file.

closed 2024-12-10 by jkrech