Open-CMSIS-Pack/cpackget #383: This command should result in error `cpackget add Keil::STM32U5xx_DFP@>=3.0.0`

ReinhardKeil, 2024-11-14:
**Describe the bug**

This command should result in error `cpackget add Keil::STM32U5xx_DFP@>=3.0.0`  

Instead it is silently accepted and no error is issued.

**Expected behavior**

Issue error, i.e. pack version incorrect format

**Environment (please complete the following information):**
 - Windows
 - CMSIS-Toolbox 2.6.1.

jkrech, 2024-11-14:
IMHO this is not a bug.

a) it is the command shell that interprets the `>` before the `cpackget` is even launched. Therefore there is no means for the tool to throw an error.
b) if you put the parameter in quotes the command works as expected:
```
cpackget add "Keil::STM32U5xx_DFP@>=3.0.0"
```

on windows the quotes are also required for the `^` as it will be removed by the command shell:
```
cpackget add "Keil::STM32U5xx_DFP@^3.0.0"
```

ReinhardKeil, 2024-11-15:
@jkrech you are correct. I overlooked that `>` is the pipe command

closed 2024-11-15 by ReinhardKeil