Open-CMSIS-Pack/cpackget #385: [cpackget] Unable to download a pack file that contains the build identifier

Torbjorn-Svensson, 2024-11-19:
**Describe the bug**

Imagine a pack index file that contains a pack and the pack file contains the build identifier. Example:
``` xml
<pdsc url="https://www.keil.com/pack/" vendor="ARM" name="CMSIS" version="6.1.0+foo" />
```

Then we try to install it using cpackget:

```
cpackget add --verbose ARM::CMSIS@6.1.0+foo
``` 

The command would generate output that ends with:

```
D: bad status: 404
E: "https://www.keil.com/pack/ARM.CMSIS.6.1.0.pack": bad request
```

But it should try to download `https://www.keil.com/pack/ARM.CMSIS.6.1.0+foo.pack` instead.
In the .pdsc file embedded in the .pack file, the the release node is:
```xml
  <releases>
    <release date="2024-02-30" version="6.1.0+foo">
      the change log entries...
   </release>
  </releases>
```

**To Reproduce**
See above.

**Expected behavior**
The pack file should be downloaded.

**Environment (please complete the following information):**
 - OS: any
 - Version 2.1.4?

**Additional context**
Obviously, we did not see the problem with the ARM.CMSIS.6.1.0 pack, but an internal pack. The steps above should highlight the question regardless of the pack name.
The .pack files is named including the metadata/build identifier on the server.

jkrech, 2024-11-26:
When installing a pack file from disk with meta information in the version:
`ARM.CMSIS.6.0.0-dev35+geb1d42a.pack`
the file will be stored in `.Download` without meta information
`${CMSIS_PACK_ROOT}/.Download/ARM.CMSIS.6.0.0-dev35.pack`
the content of the pack version will be extracted into a version folder without meta information
`${CMSIS_PACK_ROOT}/ARM/CMSIS/6.0.0-dev35/`

The pdsc file contains the meta information in the `<release>` attribute
`<release version="6.0.0-dev35+geb1d42a">`
however the tools reading pdsc files must ignore the meta information.

`*.cbuild-pack.yml`:
```
  resolved-packs:
    - resolved-pack: ARM::CMSIS@6.0.0-dev35
      selected-by-pack:
        - ARM::CMSIS@6.0.0-dev35+geb1d42a
```

slhultgren, 2025-01-10:
Hmm. so just to be clear, do we want it to be like this then?
1. Index can contain buildidentifier `(*.pidx)`
2. Server file name can contain buildidentifier `(https://packs.example.com/example.pack.1.2.3+example.pack)`
3. Local cache file name can contain buildidentifier `($CMSIS_PACK_ROOT/.Download/example.pack.1.2.3+example.pack)`
4. Local installed folder name can contain buildidentifier `($CMSIS_PACK_ROOT/example/pack/1.2.3+example/)`
5. *.yml files can contain buildidentifier `(csolution.yml, cproject.yml, cbuild-pack.yml, cbuild.yml, ...others?)`

jkrech, 2025-01-24:
1. Index can contain buildidentifier `(*.pidx)` 
2. Pack filename on server can contain buildidentifier `(https://packs.example.com/example.pack.1.2.3+example.pack)`
3. Local cache pack file name must not contain buildidentifier `($CMSIS_PACK_ROOT/.Download/example.pack.1.2.3+example.pack)`
4. Local installed folder name must not contain buildidentifier `($CMSIS_PACK_ROOT/example/pack/1.2.3+example/)`
5. `*.cbuild-pack.yml` output file could contain buildidentifier, however this information would be ignored when checking whether the right version is installed.

PR #457 (2025-02-21): #385, show error message if a packID to be added contains metadata
## Fixes
#385 

## Changes

## Checklist

- [x] 🤖 This change is covered by unit tests (if applicable).
- [x] 🤹 Manual testing has been performed (if necessary).
- [x] 🛡️ Security impacts have been considered (if relevant).
- [x] 📖 Documentation updates are complete (if required).
- [ ] 🧠 Third-party dependencies and TPIP updated (if required).
commit cbb44196 (2025-02-21): #385, show error message if a packID to be added contains metadata (#457)

## Fixes
#385 

## Changes

## Checklist

- [x] 🤖 This change is covered by unit tests (if applicable).
- [x] 🤹 Manual testing has been performed (if necessary).
- [x] 🛡️ Security impacts have been considered (if relevant).
- [x] 📖 Documentation updates are complete (if required).
- [ ] 🧠 Third-party dependencies and TPIP updated (if required).

---------
closed 2025-02-21 by bgn42