Open-CMSIS-Pack/cpackget #436: [cpackget] init command enhancement to use local pidx file

minicarpet, 2025-02-02:
**cpackget init command cannot use local file**
Only online file can be used. I would like to have a way to use local file, indeed, I have vidx file which use non public pidx file from my organization and keil pidx file from : https://www.keil.com/pack/Keil.pidx to have official package and organization package.

exemple of vidx file:
```
<index schemaVersion="1.0" xmlns:xs="http://www.w3.org/2001/XMLSchema-instance" xs:noNamespaceSchemaLocation="PackIndex.xsd">
  <vendor>MyVendor</vendor>
  <url></url>
  <timestamp>2024-09-30T06:30:00</timestamp>
  <vindex>
    <pidx vendor="Keil" url="https://www.keil.com/pack/"/>
    <pidx vendor="MyVendor" url="https://artifactory.myvendor.com/pack/"/>
  </vindex>
</index>
```
were https://artifactory.myvendor.com/pack/ store all provided non public package (means pdsc file and pack files under a folder, ie: https://artifactory.myvendor.com/pack/vendor_pack/)
So I expect my index.pidx having this line compare to keil index.pidx
```
<pdsc vendor="Hexagon" url="https://artifactory.myvendor.com/vendor_pack/" name="vendor_pack" version="0.1.2" />
```

A solution could be to add an other pidx file to .Web folder or append index.pidx file downloaded from keil with pdsc xml attribute.

jkrech, 2025-02-03:
The `.Web` folder is intended for public packs listed in an index.pidx (e.g. from Arm). However you can compose your own `index.pidx` which you register using `cpackget init <index-url>`. Note that the <url> tag inside the index.pidx is used to download updates of the registry. You may use the `vidx2pidx` utility to generate an index.pidx from multiple vidx files.
If your vidx is public and of general interest, we could add it to the import.

Alternatively if you are adding a pack manually once, the <url> will be checked for updates when `update-index` is executed. Basically the latest copy of the pdsc of such packs are stored in `.Local`.

minicarpet, 2025-02-03:
This is what I do for now, using `vidx2pidx` tool with the shown vidx file. However, I will have to generete it sometims to having last package (if new package of MDK::Middleware is available for example) I would like to avoid this.

Our package is attended to bu used with our organization board and protocol, no interest to having public.

So it would be interesting to have `cpackget init` using vidx file or using multiple pidx files ?

I will try out to check what is going on when installing manually in that case.

jkrech, 2025-04-01:
@minicarpet is there any update on this topic from your side?

closed 2025-04-16 by jkrech

minicarpet, 2025-04-17:
I tried to download and install manually my pack using `cpackget add my_pack.pack --agree-embedded-license` and this work fine.