Open-CMSIS-Pack/cpackget #499: [Bug]: cpackget add ARM::ethos-u-core-driver@1.24.8 fails

ReinhardKeil, 2025-04-01:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

It appears that cpackget is case sensitive on pack names.

This causes a problem when a project defines --pack: ARM::ethos-u-core-driver@1.24.8
The missing pack is not found using 

cpackget add ARM::ethos-u-core-driver@1.24.8

### Expected Behavior

Pack names should be case insensitive

### Steps To Reproduce

cpackget add ARM::ethos-u-core-driver@1.24.8

### Affected tools

- [ ] cbuild
- [ ] cbuild2cmake
- [x] cpackget
- [ ] csolution
- [ ] packchk
- [ ] svdconv

### Affected operating systems

- [x] Windows amd64
- [ ] Windows arm64
- [ ] Linux amd64
- [ ] Linux arm64
- [ ] MacOS amd64
- [ ] MacOS arm64

### Environment

cpackget v2.1.6 on Windows 11

ReinhardKeil, 2025-04-01:
IMHO we should always search case insensitive.

jkrech, 2025-04-14:
Fix can be validated in a nightly build of CMSIS-Toolbox (2025-04-10 or later)

jkrech, 2025-04-16:
Works as expected.

closed 2025-04-16 by jkrech