Open-CMSIS-Pack/cpackget #609: [Bug]: Download Error on Debian for Infineon.T2G-B-H_DFP.1.2.1.pack

ReinhardKeil, 2025-09-30:
### Is there an existing issue for this?

- [x] I have searched the existing issues

### Current Behavior

`cpackget add Infineon::T2G-B-H_DFP@1.2.1` on Debian Linux results in:

```
I: Adding pack "Infineon::T2G-B-H_DFP@1.2.1" 
E: Get "https://itools.infineon.com/cmsis_packs/T2G-B-H/Infineon.T2G-B-H_DFP.1.2.1.pack": tls: failed to verify certificate: x509: certificate signed by unknown authority 
E: "https://itools.infineon.com/cmsis_packs/T2G-B-H/Infineon.T2G-B-H_DFP.1.2.1.pack": failed to download file 
Error: Process completed with exit code 255.
```

Other packs (i.e. Arm:CMSIS@6.2.0) don't show this error.

wget outputs a similar message, which indicates that something with the certificate is wrong. However there is no error on MacOS, Windows, or GitHub hosted runners using Ubuntu.  

```
wget -P Downloads/ https://itools.infineon.com/cmsis_packs/T2G-B-H/Infineon.T2G-B-H_DFP.1.2.1.pack
--2025-09-29 17:13:59--  https://itools.infineon.com/cmsis_packs/T2G-B-H/Infineon.T2G-B-H_DFP.1.2.1.pack
Resolving itools.infineon.com (itools.infineon.com)... 217.10.52.14
Connecting to itools.infineon.com (itools.infineon.com)|217.10.52.14|:443... connected.
ERROR: The certificate of ‘itools.infineon.com’ is not trusted.
ERROR: The certificate of ‘itools.infineon.com’ doesn't have a known issuer.
```

Raspberry Pi System Info:

```
>hostnamectl
Static hostname: devlinux
       Icon name: computer
      Machine ID: 51d83b8784934cf1ba018a140dae8249
         Boot ID: 07de96fd8a5c4597b289d8bf7b630af9
Operating System: Debian GNU/Linux 12 (bookworm)
          Kernel: Linux 6.12.20+rpt-rpi-2712
    Architecture: arm64
```

Questions/Actions:
- provide more information on what is wrong with the certificate
- add option to `cpackget` that allows to bypass certificate test.  `wget` seems to have such an option

### Affected tools

- [ ] cbuild
- [ ] cbuild2cmake
- [ ] cpackget
- [ ] csolution
- [ ] packchk
- [ ] svdconv

### Affected operating systems

- [ ] Windows amd64
- [ ] Windows arm64
- [ ] Linux amd64
- [ ] Linux arm64
- [ ] MacOS amd64
- [ ] MacOS arm64

jkrech, 2026-02-06:
As we have already implemented a flag to avoid the check, we need to test whether it works rather than suggesting to reject the change made.

Could you please install the old Debian version again? Or is that too much effort?
If you find another way to validate the fix that would be great.

@bgn42 Bernd, how did you test this?

Evagan2018, 2026-02-09:
I have installed a recent version (nightly build) of the cmsis-toolbox.

The cmsis-toolbox-linux-arm64 has been used for the Raspberry Pi.
Link to artifact: https://github.com/Open-CMSIS-Pack/cmsis-toolbox/actions/runs/21810963025/artifacts/5426488258

**Prework**
For downloading an artifact you need to authenticate with your github token.

TOKEN='<YOUR_GITHUB_TOKEN>'

Download artifact by using github token:
curl -L \
  -H "Authorization: token $TOKEN" \
  -H "Accept: application/vnd.github+json" \
  -H "User-Agent: raspi" \
  -o cmsis-toolbox-linux-arm64.zip \
  https://api.github.com/repos/Open-CMSIS-Pack/cmsis-toolbox/actions/artifacts/5426488258/zip

**Testing the cpackget command without flag**
cpackget add Infineon::T2G-B-H_DFP@1.2.1
I: Adding pack "Infineon::T2G-B-H_DFP@1.2.1"
E: Get "https://itools.infineon.com/cmsis_packs/T2G-B-H/Infineon.T2G-B-H_DFP.1.2.1.pack": tls: failed to verify certificate: x509: certificate signed by unknown authority
E: "https://itools.infineon.com/cmsis_packs/T2G-B-H/Infineon.T2G-B-H_DFP.1.2.1.pack": failed to download file

**Testing the cpackget command with the flag --insecure-skip-verify**
cpackget add Infineon::T2G-B-H_DFP@1.2.1 --insecure-skip-verify
I: Adding pack "Infineon::T2G-B-H_DFP@1.2.1"
I: Downloading Infineon.T2G-B-H_DFP.1.2.1.pack...
I: 100% |█████.........██████████████| (33/33 MB, 6.6 MB/s)
I: Extracting files to /home/devuser/packs/Infineon/T2G-B-H_DFP/1.2.1...
I: 100% |█████........ █████████████| (7482/7482, 918 it/s)

I can confirm that using the flag --insecure-skip-verify allows the download of this pack.

So, I think we can close this bug.